shadow578/Marlin-H32 #12: [BUG] (bug summary)

shadow578, 2023-03-10:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

a

### Version of Marlin Firmware

a

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-04-01:
you know who would know better? @Mriscoc

classicrocker883, 2023-04-22:
is there anything else regarding Hc32f46x and PlatformIO? any other issues or something that needs to be addressed and worked on?

shadow578, 2023-04-26:
I'm currently working on updating the arduino core with a more recent version of the DDL, since the current one is the one supplied (and modified) by voxelab. 
That'll take a bit of time, as i've to wait on hdsc and theres a substantial timezone difference...

Other than that, it should just need testing if Marlin (& meaningful Marlin features) work as expected. Stuff like BLTouch. 
Since HC32 dev-boards are not available anyway, i don't think it makes sense to go for a general purpose arduino core like for the STM32.

shadow578, 2023-04-26:
+ mid-term, it'd be nice to test other printers that use the HC32F460. 

But i'd really prefer running on a clean DDL first before going that route.

classicrocker883, 2023-04-28:
sounds like a plan. im sure youd be able to make your own dev board by swapping from a STM32. i browsed through the datasheet a bit and compared some of the pinouts with the stm32/gd32, and hc32 is basically just another clone. however, the *pins*.h file showing the pinouts like in the **pins_CREALITY_V4.h** show a slight difference but i cant see why. I had always thought an HC32 chip could be swapped with STM32 like GD32 and N32, even though all the pins match, there are still some differences, but surely it can work fine with a dev board.

anyway having an updated core library would be great, as long as it works and is compatible with everything else.

classicrocker883, 2023-05-15:
so i was messing around and enabled the *Device Driver Library* (this is the ddl you mentioned right?) by uncommenting `USE_DEVICE_DRIVER_LIB` in **hc32_ddl.h** file. everything compiled fine without error, but I haven't gone to test and see if it actually works, or what difference it made. just thought you would know, and the changes i had to make were

say for example 
```
#if (DDL_CMP_ENABLE == DDL_ON)
#include "hc32f46x_cmp.h"
#endif /* DDL_CMP_ENABLE */
```
just changed to `#include "../library/inc/hc32f46x_cmp.h"`
could not locate **hc32f46x_usbfs.h**

and went into **ddl_config.h** to manually define them all to `DDL_ON`