shadow578/Marlin-H32 #14: [BUG] (bug summary)

classicrocker883, 2023-03-31:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

H32.mk:121: *** target pattern contains no `%'.  Stop.

### Version of Marlin Firmware

2.1.3

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

shadow578, 2023-04-04:
do i understand correctly that you've (at first) tried compiling under windows? if thats the case, that'll not work since the makefile uses linux commands internally. You __have to__ use linux (or WSL). 

as for the part with removing `-Wl,--no-warn-rwx-segment`, are you sure you've installed the latest gcc toolchain for arm-none-eabi? 
could you run `arm-none-eabi-gcc --version` and `arm-none-eabi-g++ --version` and post here?

classicrocker883, 2023-04-04:
>/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: unrecognized option '--no-warn-rwx-segment'

arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
arm-none-eabi-g++ (15:10.3-2021.07-4) 10.3.1 20210621 (release)

classicrocker883, 2023-04-04:
I am having issue with `BLTOUCH` enabled. it gives me an error of undefined reference to `libServo::***` with different functions of the libServo class in Servo.h and Servo.cpp in HAL/STM32F1

classicrocker883, 2023-04-05:
this issue  -Wl,--no-warn-rwx-segment has somehow gone away, i can leave it in the .mk file

closed 2023-04-05 by classicrocker883

shadow578, 2023-04-07:
> arm-none-eabi-gcc (15:10.3-2021.07-4) 10.3.1 20210621 (release)
> arm-none-eabi-g++ (15:10.3-2021.07-4) 10.3.1 20210621 (release)

you're using a fairly old version of the arm-none-eabi toolchain. 
please take a look at https://github.com/shadow578/Marlin-H32/blob/aquila/.toolchain/README.md for instructions to set up a more recent one

shadow578, 2023-04-07:
BLTouch is currently not working, as also described in #15 .

classicrocker883, 2023-04-08:
I thought that there was a newer version! I've been trying to update automatically it without having to do all the manual BS, the voxelab official had the v10.xx linked and I just went with that. if I search Google for arm-none-eabi-g++  the v10 is one of the first choices, I wonder why it doesn't link to the newer version.

classicrocker883, 2023-04-08:
i am still receiving the `-Wl,--no-warn-rwx-segment \`

what I had to do to get the v12.2 of arm-none-eabi-g** was add 
_`deb http://archive.ubuntu.com/ubuntu lunar main universe`_
_`deb http://archive.ubuntu.com/ubuntu lunar-updates main universe`_
to the **/etc/apt/sources. List** because im still on ubuntu LTS 22.04

i do not know how to install the tar.xz file by manually downloading it, i tried following the directions, there is no ./configure or make file. so do i just unzip it and put it wherever and just echo a path to it? anyway i was able to do it the automatic update way which is way preferred. 

still getting that unrecognized option

>Linking Firmware
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: unrecognized option '--no-warn-rwx-segment'
/usr/lib/gcc/arm-none-eabi/12.2.1/../../../arm-none-eabi/bin/ld: use the --help option for usage information
collect2: error: ld returned 1 exit status
make: *** [H32.mk:175: /home/andrew/Marlin-H32-3/Marlin/build/firmware. Elf] Error 1