shadow578/Marlin-H32 #16: [BUG] compiling using PlatformIO?

classicrocker883, 2023-04-05:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

is the platformIO builder ready for this?
when i put this in the platformio.ini this error pops up

>Processing HC32F46x_AQUILA_X2 (platform: hc32f46x; board: hc32f46x; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Platform Manager: Installing hc32f46x
Error: Could not find the package with 'hc32f46x' requirements for your system 'linux_x86_64'

it also said `for your system 'windows...` when trying in the windows env.

### Version of Marlin Firmware

pio-build

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

shadow578, 2023-04-07:
nope, platformio build is currently __not__ ready, as it would require creating platform, framework, and toolchain packages for the hc32 platform. see #12 for some more details. 
if you wish to help with implementing those packages, i can give you access to some internal ones i have. tho be warned they are *very* unstable.

closed 2023-04-07 by shadow578

classicrocker883, 2023-04-08:
> nope, platformio build is currently **not** ready, as it would require creating platform, framework, and toolchain packages for the hc32 platform. see #12 for some more details. if you wish to help [...]

yeah I'm down, I'll try to help out as best I can. if you be so kind as to let me know any steps to take. or do you have a branch I can clone and I'll pick it up from there?