shadow578/Marlin-H32 #18: [BUG] Trying to get it work with 12864 (Aquila C2 / CR10 / Ender-2 stock display)

classicrocker883, 2023-04-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

after installing the `U8glib-HAL` library through platformio (by adding to project) i was able to stop the "cannot find <U8glib-HAL.h> " errors and it almost compiled successful until it was just about to link the firmware and gave me all kinds of undefined reference errors. not sure why. maybe this is a step closer?

>Compile lib/h32_core/hdsc/common/startup_hc32f46x.S
Linking Firmware
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/MarlinCore.cpp.o: in function `setup':
/home/andrew/Marlin-H32-3/Marlin/src/MarlinCore.cpp:1338: undefined reference to `MarlinUI::show_bootscreen()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/src/MarlinCore.cpp:1625: undefined reference to `MarlinUI::bootscreen_completion(unsigned long)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::set_contrast(unsigned char)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:110: undefined reference to `MarlinUI::_set_contrast()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `U8GLIB::cbegin()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/dogm/../../../../U8glib-HAL/src/U8glib-HAL.h:56: undefined reference to `u8g_Begin'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `U8GLIB::firstPage()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/dogm/../../../../U8glib-HAL/src/U8glib-HAL.h:115: undefined reference to `u8g_FirstPage'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::update()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:1163: undefined reference to `MarlinUI::set_font(MarlinFont)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `U8GLIB::setColorIndex(unsigned char)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/dogm/../../../../U8glib-HAL/src/U8glib-HAL.h:129: undefined reference to `u8g_SetColorIndex'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `U8GLIB::nextPage()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/dogm/../../../../U8glib-HAL/src/U8glib-HAL.h:116: undefined reference to `u8g_NextPage'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::update()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:1234: undefined reference to `u8g'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:1234: undefined reference to `u8g'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:1222: undefined reference to `MarlinUI::clear_lcd()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::init()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:221: undefined reference to `MarlinUI::init_lcd()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::reinit_lcd()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.h:207: undefined reference to `MarlinUI::init_lcd()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::status_screen()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:720: undefined reference to `MarlinUI::draw_status_screen()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::kill_screen(__FlashStringHelper const*, __FlashStringHelper const*)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.cpp:737: undefined reference to `MarlinUI::draw_kill_screen()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/marlinui.cpp.o: in function `MarlinUI::reinit_lcd()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/marlinui.h:207: undefined reference to `MarlinUI::init_lcd()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu.cpp.o: in function `MarlinUI::synchronize(__FlashStringHelper const*)::{lambda()#1}::_FUN()':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu.cpp:239: undefined reference to `MenuItem_static::draw(unsigned char, __FlashStringHelper const*, unsigned char, char const*)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu.cpp.o: in function `MarlinUI::goto_screen(void (*)(), unsigned short, unsigned char, unsigned char)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu.cpp:209: undefined reference to `MarlinUI::clear_lcd()'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu.cpp.o: in function `MenuEditItemBase::draw_edit_screen(char const*)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu.h:190: undefined reference to `MenuEditItemBase::draw_edit_screen(__FlashStringHelper const*, char const*)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu.cpp.o: in function `MenuItem_confirm::select_screen(__FlashStringHelper const*, __FlashStringHelper const*, void (*)(), void (*)(), __FlashStringHelper const*, char const*, __FlashStringHelper const*)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu.cpp:368: undefined reference to `MenuItem_confirm::draw_select_screen(__FlashStringHelper const*, __FlashStringHelper const*, bool, __FlashStringHelper const*, char const*, __FlashStringHelper const*)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu_advanced.cpp.o: in function `MenuItem_back::draw(bool, unsigned char, __FlashStringHelper const*)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu.h:85: undefined reference to `MenuItemBase::_draw(bool, unsigned char, __FlashStringHelper const*, char, char)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu_advanced.cpp.o: in function `TMenuEditItem<MenuEditItemInfo_float42_52>::draw(bool, unsigned char, __FlashStringHelper const*, float*, ...)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu_item.h:88: undefined reference to `MenuEditItemBase::draw(bool, unsigned char, __FlashStringHelper const*, char const*, bool)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu_advanced.cpp.o: in function `MenuItem_back::draw(bool, unsigned char, __FlashStringHelper const*)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu.h:85: undefined reference to `MenuItemBase::_draw(bool, unsigned char, __FlashStringHelper const*, char, char)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu_advanced.cpp.o: in function `TMenuEditItem<MenuEditItemInfo_float4>::draw(bool, unsigned char, __FlashStringHelper const*, float*, ...)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu_item.h:88: undefined reference to `MenuEditItemBase::draw(bool, unsigned char, __FlashStringHelper const*, char const*, bool)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/lcd/menu/menu_advanced.cpp.o: in function `MenuEditItemBase::draw(bool, unsigned char, __FlashStringHelper const*, __FlashStringHelper const*)':
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu.h:182: undefined reference to `MenuEditItemBase::draw(bool, unsigned char, __FlashStringHelper const*, char const*, bool)'
[... cut ...]
/home/andrew/Marlin-H32-3/Marlin/src/lcd/menu/menu_item.h:42: undefined reference to `MenuItemBase::_draw(bool, unsigned char, __FlashStringHelper const*, char, char)'
/usr/lib/gcc/arm-none-eabi/10.3.1/../../../arm-none-eabi/bin/ld: /home/andrew/Marlin-H32-3/Marlin/build/src/sd/cardreader.cpp.o: in function `CardReader::manage_media()':
/home/andrew/Marlin-H32-3/Marlin/src/sd/cardreader.cpp:491: undefined reference to `MarlinUI::detected()'

### Version of Marlin Firmware

current

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

shadow578, 2023-04-07:
adding any library to platformio will __100% not work__, as the project is currently not being build using platformio (unless you did something to change that, ofc). 

In case you actually did add all the required libraries, the error you're getting could be caused because most of the lcd source files are not actually compiled by default (to safe time when compiling). 
You can change this by removing the `src/lcd/***` lines in H32.mk (https://github.com/shadow578/Marlin-H32/blob/291956a5e6275d08517c3b3cad2f89a93bc6f2a8/Marlin/H32.mk#L60).

closed 2023-04-14 by shadow578

classicrocker883, 2023-04-14:
just FYI, although this is a step closer, I don't believe I was able to get it to compile fully having `CR10_STOCKDISPLAY` for the lcd screen.

classicrocker883, 2023-04-16:
could we reopen this issue?

so now that platformio: build works, im trying to get this to compile using `CR10-STOCKDISPLAY` instead of `DWIN_LCD_*`

what i had to do was simply add the code from **src/pins/pins_CREALITY_V4.h** to **pins_Aquila_X2.h**
[like so --> pins_Aquila_X2.txt](https://github.com/shadow578/Marlin-H32/files/11241283/pins_Aquila_X2.txt)

in this file **.platformio\packages\framework-arduino-hc32f46x\cores\arduino\Printable.h** its giving undefined reference to _**class**_. apparently since it compiles in C, "*class*" is a C++ thing so i wonder if there is some way around this. some files like **Printable.h** in **\.platformio\packages\framework-arduino-hc32f46x\** (which gets downloaded from your other repository) are exactly the same the other compiler libraries which make a successful firmware build using normal 3d printer parameters like creality ender 3 - which use STMF1.

besides the undefined reference to a **_class_**, it gives another error having a second defined function

>wirish_math.h:46:17: error: conflicting types for 'makeWord'
wirish_math.h:10:6: error: conflicting types for 'random'

so what needs to be done is to exclude that particular function from that specific library. ive been able to add something like this to **/ini/h32.ini**
>Wl,-L.platformio\packages\toolchain-gccarmnoneeabi\arm-none-eabi\lib -lcstdlib,--exclude-symbols=random

this code doesn't exactly work, its based on my research and references (like ChatGPT in VSCode). its very close to what needs to get done.

if it can exclude `long random (void);` from <stdlib>, probably from gccarmnoneeabi since its defined in the hc32f46x framework, that might work and its a step in the right direction.