shadow578/Marlin-H32 #24: [BUG] Bootloop when compiling with BLTouch support enabled

pipe01, 2023-04-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

If I compile the latest code with the recommended configuration files (adjusted for some Marlin updates), my Aquila X2 runs perfectly, but of course can't use the BLTouch. If I enable the corresponding #defines for enabling the probe (`BLTOUCH`, `AUTO_BED_LEVELING_BILINEAR` and `Z_SAFE_HOMING`) and run this, my aquila just loops the startup animation.

[configuration.zip](https://github.com/shadow578/Marlin-H32/files/11214167/configuration.zip)

### Version of Marlin Firmware

Latest commit

### Printer model

Voxelab Aquila X2 H32

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-04-12:
@double-brown had this same issue. i do know that the **pio_build/bugfix** branch allows it to compile using platformio: build. have you tried that branch instead?

shadow578, 2023-04-13:
hi  @pipe01 , as already suggested please try the `pio_build/bugfix` branch (https://github.com/shadow578/Marlin-H32/tree/pio_build/bugfix). 
Please keep in mind that that branch no longer uses the makefile-based build system, but platformio. See [here](https://github.com/shadow578/Marlin-H32/tree/pio_build/bugfix#building) for details.

pipe01, 2023-04-13:
Thanks for the replies, but unfortunately I get the same issue with that branch.

pipe01, 2023-04-13:
Still no luck I'm afraid.

```
start
 Watchdog Reset
Marlin bugfix-2.1.x
echo: Last Updated: 2023-04-07 | Author: (shadow578, Aquila X2 (H32))
echo: Compiled: Apr 13 2023
echo: Free Memory: 176451  PlannerBufferBytes: 3200
echo:V87 stored settings retrieved (614 bytes; crc 10416)
BL24CXX Check succeeded!
version 1.2

UID =(0x43323135,0xFF025030,0xFFFF5A4F)
sdio init fail!
Disk init

Tips ------ None Factory file

Tips ------ None Firmware file

GotoApp->addr=0xC000
```

classicrocker883, 2023-04-13:
try to enable or disable `#define MACHINE_UUID` (which ever isn't currently set)
enable Meanpack
how are you reading this? through octoprint? or via usb? 

take a look at these configs. compare them with yours and enable or disable what you need to so they more or less match. are you using JyersUI? this is the only one apparently working for now with the original repo. 

[Configuration.h](https://raw.githubusercontent.com/classicrocker883/Marlin-HC32/pio_build/bugfix/Marlin/Configuration.h)

[Configuration_adv.h](https://raw.githubusercontent.com/classicrocker883/Marlin-HC32/pio_build/bugfix/Marlin/Configuration_adv.h)

pipe01, 2023-04-13:
> try to enable or disable #define MACHINE_UUID (which ever isn't currently set)
> enable Meatpack in _adv.h
> edit: in src/module/planner.h change PlannerBlock to block_t

Doesn't seem to work either.

> how are you reading this? through octoprint? or via usb?

Just through USB, `cat /dev/ttyUSB0`.

> take a look at these configs. compare them with yours and enable or disable what you need to so they more or less match. 

There are a ton of differences, I tried just replacing my configs with those and it doesn't work either.

> are you using JyersUI? this is the only one apparently working for now with the original repo.

Not sure what that is, is it for the display?

Thanks for troubleshooting with me, I really want to get this working.

classicrocker883, 2023-04-13:
how are you uploading the firmware? u should be putting it on a Micro SD card, in a folder called firmware right?

as for display there are several options in the Configuration.h file, near the end of it look for DWIN_LCD_JYERSUI

make sure the other dwin lcd aren't defined by using // infront of the line.

so youve compiled everything correctly and it was successful? and does it flash OK and u can use the lcd screen?

pipe01, 2023-04-13:
> how are you uploading the firmware? u should be putting it on a Micro SD card, in a folder called firmware right?

Yup.

> as for display there are several options in the Configuration.h file, near the end of it look for DWIN_LCD_JYERSUI
> make sure the other dwin lcd aren't defined by using // infront of the line.

I see. I'm not too worried about the display right now though as I don't really use it.

> so youve compiled everything correctly and it was successful? and does it flash OK and u can use the lcd screen?

It compiles but it still bootloops. I have noticed though that the reset reason is "Watchdog reset", so maybe there's some infinite loop or something somewhere? I recorded the terminal so you can see the timing of the log entries: https://asciinema.org/a/wLoxTbKV66WxZNVj0PVLu5RuX

pipe01, 2023-04-13:
I enabled

pipe01, 2023-04-13:
I enabled MARLIN_DEV_MODE and took a serial log from the start of the update process to the bootloop: https://pastebin.com/26Ubx2my

classicrocker883, 2023-04-13:
if it bootloops I would say that is the issue. you want it so that doesn't happen. the lcd screen should be usable even if u don't intend to use it. u should be able to enable DWIN_LCD_JYERSUI and everything should work as normal

classicrocker883, 2023-04-13:
oh the BLTouch may not be working. I'm actually unsure if it does work because there was an issue before. u can try using the Manual Mesh settings instead and see if everything works as it should, just without a probe, that's what I would do.

pipe01, 2023-04-13:
It seems like the last line that gets output is `echo:[2242] DWIN_InitScreen()` before restarting, so it seems like it has something to do with that code. That's very weird though, because if I disable `BLTOUCH` the screen and everything else works just fine.

pipe01, 2023-04-13:
I disabled all the `DWIN_` defines and `SDCARD_SORT_ALPHA` and there is no more bootlooping, so it is definitely related to that.

classicrocker883, 2023-04-13:
that is some good info to know. so you're not using the screen at all? and sdcard_sort_alpha definitely stopped the bootloops?

pipe01, 2023-04-13:
Actually it seems like enabling `DWIN_CREALITY_LCD` is what causes the bootloops, even with the SD card disabled.

pipe01, 2023-04-13:
`DWIN_CREALITY_LCD_JYERSUI` does seem to work, so I guess I'll be using that. Do you know where I can find the corresponding `DWIN_SET` files?

classicrocker883, 2023-04-14:
yes u can use https://github.com/alexqzd/Marlin/tree/main/Display%20firmware/Firmware%20Sets

or any other DWIN_SET it doesn't have to be reflashed if you already done so.

oh another thing for the display, you may need to enable
`#define USE_STOCK_DWIN_SET`, which is located in src/lcd/e3v2/common/dwin_set.h

pipe01, 2023-04-14:
Oh I didn't know about that define, thank you for the help!

pipe01, 2023-04-14:
That seems to be it. I imagine there must be something else wrong though, since without `BLTOUCH` enabled `DWIN_CREALITY_LCD` works fine.

classicrocker883, 2023-04-14:
> That seems to be it. I imagine there must be something else wrong though, since without `BLTOUCH` enabled `DWIN_CREALITY_LCD` works fine.

how about `DWIN_CREALITY_LCD_JYERSUI`, is that the same issue, having **BLTouch** enabled cause it to not work, or no difference?

pipe01, 2023-04-14:
With JyersUI enabled it does work, even with `BLTOUCH` enabled.

MadmanPan, 2023-05-02:
https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3-hc could I test this firmware? I have the same issue that BL touch loops the startup animation. Thx.

MadmanPan, 2023-05-02:
> 启用 JyersUI 后，即使启用，它也可以工作。`BLTOUCH`
https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3-hc could I test this firmware? I have the same issue that BL touch loops the startup animation. Thx.

classicrocker883, 2023-05-10:
adding `BOOTSCREEN_TIMEOUT 1100` to the **dwin.cpp** file under the function that `init()` initializes the bootscreen. this is true for the ProUI, not sure for JyersUI or other. but this fix seems to have worked.