shadow578/Marlin-H32 #25: [BUG] flash memory limit low?

classicrocker883, 2023-04-12:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

it seem as though **FLASH:** is limited to 211936 B. the other Aquila chips are limited to 89% of 262144 Bytes (= 256kb) or ~233,300 B.

unlocking this extra space would be helpful. is there a way the chip can handle more flash space? or would this ultimately brick it?

### Version of Marlin Firmware

pio_build/bugfix

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

shadow578, 2023-04-13:
could you give more details on what branch you're working with? Is it the `aquila` branch or one of the `pio` branches?

Answering more generall (as this is probably is true for both):

1. the HC32F46x series consists of two main ICs: the `HC32F460xCxx` and `HC32F460xExx` (were x=dont care), with the 
main difference beign that the first has 256K of flash, and the former has 512K. 

2. not all of flash is usable. from `0x0000` to `0xC000` is the bootloader, so of the 256K version only (262.144 - 0xC000 =) 212.992 (about 208K) are usable for Marlin. This ofc assumes one wants to keep the bootloader, if you were to directly flash the ic you cold use all of flash.

So, TL;DR:

no, you cannot unlock more flash unless you remove the bootloader, which you should probably not do.
However, you could check if your board uses the 512K version of the HC32F46x. If this is the case, you can use more flash.

classicrocker883, 2023-04-13:
that is a good reply thank you. I was using the pio_buid/bugfix. so that makes me wonder what is actually on these boards, because all the G32 and N32 for Aquila are RCT6 (256k) so I imagine same goes for the H32. but is it possible they might have been shipped with the 512k version?

and what chip is on your board if I may ask. I don't actually have a Hc32, instead I've been using a G32.

shadow578, 2023-04-14:
> and how much chip memory s on your board if I may ask. 256 or 512?

i actually never checked, i always assumed they'd ship the 256K version since it's cheaper.

closed 2023-04-14 by shadow578