shadow578/Marlin-H32 #29: [BUG] (bug summary)

classicrocker883, 2023-04-18:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

this isn't particular to your repository, but is relatable  I was wondering if you could look through this code anyway. 
So I went ahead and merged the current Marlin repo with this, and Mriscoc's ProUI - wasn't easy - but it compiles fine and its all good and everything, just wanted a second opinion

so here in this file 
**[src/module/stepper.cpp](https://github.com/classicrocker883/MriscocProUI/blob/hc32_build/Marlin/src/module/stepper.cpp)** (linked to my repo as stated above)

```
 #ifdef CPU_32_BIT

    // A fast processor can just do integer division
    constexpr uint32_t min_step_rate = uint32_t(STEPPER_TIMER_RATE) / HAL_TIMER_TYPE_MAX;
    return step_rate > min_step_rate ? uint32_t(STEPPER_TIMER_RATE) / step_rate : HAL_TIMER_TYPE_MAX;

  #else
```
however I get this error
>framework-arduino-hc32f46x\cores\arduino/Arduino.h:13:21: note: 'uint32_t CPU_FREQ' is not const        
   13 |     extern uint32_t CPU_FREQ;

so I changed the above code to:
```
TERN(TARGET_HC32F46x, const, constexpr) uint32_t min_step_rate = uint32_t(STEPPER_TIMER_RATE) / HAL_TIMER_TYPE_MAX;
```

I'm wondering if that is the correct way, because after it compiles fine, but also without `const` or `constexpr` (left blank), it also compiles fine.

my question is what would you do? feel free to look over the repo which I merged to the current Marlin bugfix.

### Version of Marlin Firmware

Hybrid of Marlin, Mriscoc and this "aquila" branch

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-04-18:
update -
note: not matter if `const`/`constexpr`/or (blank), it compiles successful with no changes to RAM or FLASH, not by one byte.
it just will not compile successful using only HC32 and `constexpr`.

shadow578, 2023-05-18:
this is a known issue, i'm working on a permanent solution for this in the cores.

Basically, issue is that the clock frequency for the HC32 is configured using code (similar to how the STM32* does it) and can change during the runtime of the firmware. 
So setting it to a constant value like Marlin expects it to be is not quite as simple...

classicrocker883, 2023-05-19:
hey I'm guessing the new library fixed some things? do you know if `BLTouch` can be enabled with the other lcd dwin sets like ProUI? because it would somehow only work with JyersUI. apparently UBL or BLTouch gives constant boot loop, or restarts when saving.

shadow578, 2023-10-29:
i believe this issue is since fixed, if not please re-open

closed 2023-10-29 by shadow578