shadow578/Marlin-H32 #31: Plans to eventually get this back into mainline?

meganukebmp, 2023-05-08:
### Describe the feature you want

Great work on this so far! I'm glad someone is putting in the work needed (I tried and failed myself).
Maybe a bit too early to ask, but do you plan on contributing this back into mainline Marlin when you feel its mature enough? This would be a fantastic contribution (and possibly pulling in more maintainers as a result)

Thanks for the fantastic work here.

classicrocker883, 2023-05-10:
I'd like to say it is nearly there. I have used this in my own fork and had success, however some minor changes I had to make.

using this on the Aquila with this chip definitely works, but the next step I say is getting it to work with the 12864 lcd like on the Aquila C2.

closed 2023-11-11 by shadow578

shadow578, 2023-11-11:
opened PR https://github.com/MarlinFirmware/Marlin/pull/26414 on upstream repo just now

classicrocker883, 2023-11-15:
is everything running right so far? I heard someone else using this had an issue with auto bed leveling, I think it was restarting trying to save the mesh. are there any other issues or is everything else good to go?

shadow578, 2023-11-18:
@classicrocker883 on my machine everything is working fine, as part of https://github.com/shadow578/Marlin-H32/pull/53 i've conducted some longer running stability tests (all using bilininar ABL w/ BLTouch). 
Since then, i've actually printed a bit more, all without issues