shadow578/Marlin-H32 #34: [BUG] failed to compile hc32.ini | generic_hc32f460

classicrocker883, 2023-05-17:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

does not accept `generic_hc32f460`.
```
Error: Unknown board ID
```
replacing it with `hc32f46x_generic` does work, except now this error pops up

>KeyError: 'framework-hc32f46x-ddl':
  File "C:\Users\Andrew\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 174:
    env.SConscript("$BUILD_SCRIPT")
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Andrew\.platformio\platforms\hc32f46x\builder\main.py", line 70:
    target_elf = env.BuildProgram()
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Andrew\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 60:
    env.ProcessProgramDeps()
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Andrew\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 120:
    env.BuildFrameworks(env.get("PIOFRAMEWORK"))
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Util\envs.py", line 242:
    return self.method(*nargs, **kwargs)
  File "C:\Users\Andrew\.platformio\penv\lib\site-packages\platformio\builder\tools\piobuild.py", line 331:
    SConscript(env.GetFrameworkScript(name), exports="env")
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Andrew\.platformio\platforms\hc32f46x\builder\frameworks\arduino.py", line 15:
    SConscript(build_script)
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 662:
    return method(*args, **kw)
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 598:
    return _SConscript(self.fs, *files, **subst_kw)
  File "C:\Users\Andrew\.platformio\packages\tool-scons\scons-local-4.5.2\SCons\Script\SConscript.py", line 285:
    exec(compile(scriptdata, scriptname, 'exec'), call_stack[-1].globals)
  File "C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\tools\platformio\platformio-build-arduino.py", line 61:        
    ddl_build_script = join(env.PioPlatform().get_package_dir("framework-hc32f46x-ddl"), "tools", "platformio", "platformio-build-ddl.py")
  File "C:\Users\Andrew\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 32:
    pkg = self.get_package(name)
  File "C:\Users\Andrew\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 29:
    return self.pm.get_package(spec or self.get_package_spec(name))
  File "C:\Users\Andrew\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 21:
    owner=self.packages[name].get("owner"),

### Version of Marlin Firmware

newest aquila HC32 build

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-05-17:
https://github.com/shadow578/framework-hc32f46x-ddl

doesn't seem to be loading or installing, because files with `#include <hc32_ddl.h>` cannot find that.

classicrocker883, 2023-05-17:
I figured it out!
I had to remove `C:\Users\Andrew\.platformio\platforms\hc32f46x`
also `C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x` for good measure.

classicrocker883, 2023-05-17:
can we tweak this mem usage like it was before?

```
Linking .pio/build/HC32F46x_AQUILA_X2/firmware. Elf
Memory region         Used Size  Region Size  %age Used
       FLASH_ISR:         640 B         1 KB     62.50%
       FLASH_ICG:          32 B         32 B    100.00%
           FLASH:      174536 B     211936 B     82.35%
             RAM:       10528 B       188 KB      5.47%
         RAM_RET:          0 GB    525248 KB      0.00%
Checking size .pio/build/HC32F46x_AQUILA_X2/firmware. Elf
```

```
Linking .pio\build\HC32F460xCxx_AQUILA_V101\firmware.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:      210788 B       256 KB     80.41%
             OTP:          0 GB       1020 B      0.00%
             RAM:       12976 B       188 KB      6.74%
         RET_RAM:          0 GB         4 KB      0.00%
Building .pio\build\HC32F460xCxx_AQUILA_V101\firmware.bin
```
here is what it looks like now.

I know these chips wont fill 256k, but either the new DDL library made a difference and gave more room on the chip to flash, or now its not as easy to see how much is left.

as you can see `211936 B` is the most you can use. can we put the max mem to this like it was?

classicrocker883, 2023-05-18:
hey @double-brown would you be willing and able to test out the recently updated firmware? what im most concerned about is the size of the flash memory. because before it was limited to ~208kb for compiling. now it appears the updated code gives more room. im not sure if its actually limited to 256k or some other amount. since I do not have an HC32 to test on, would u be able to compile or use a file I will post and confirm if it indeed can use more memory?

and @shadow578 if you know anything what I mean please do chime in.

double-brown, 2023-05-18:
@classicrocker883 Yip no worries. I'll get some time this weekend to give it a try.

shadow578, 2023-05-18:
Hi @classicrocker883 

> I figured it out! I had to remove `C:\Users\Andrew\.platformio\platforms\hc32f46x` also `C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x` for good measure.

This is due to the updated DDL, i've updated the name of the generic board manifest. For future Issues, please make sure you've read #26, the steps you've described solved this issue were also described there.

---

> I know these chips wont fill 256k, but either the new DDL library made a difference and gave more room on the chip to flash, or now its not as easy to see how much is left.

With the new DDL, i've changed the linker script to be much closer to the stock one provided by HUADA. 
the `FLASH` region now also includes `ISR` (= the VOT), `ICG` (= initial config), and the bootloader offset. 

Changing the linker script back to how it was before will not happen since 
a) keeping the linker script closer to stock makes updating the DDL way easier
b) considering the DDL platform as a generic SDK for the HC32F640 instead of just using it for Marlin, outputting the entire flash size makes more sense (non-marlin wouldn't have a bootloader)

> what im most concerned about is the size of the flash memory. because before it was limited to ~208kb for compiling. now it appears the updated code gives more room. im not sure if its actually [...]

this is not correct, there are still only 208K available to marlin. 
Only now the size of the `FLASH` region is simply the equal to the entire flash of the device, and the used flash % includes the bootloader (~ 48K) and the entire binary.

closed 2023-05-18 by shadow578