shadow578/Marlin-H32 #35: [FR] Having support for C2 - 12864 LCD

classicrocker883, 2023-05-28:
### Is your feature request related to a problem? Please describe.

Yes, `  CR10_STOCKDISPLAY  ` is what these lcd controllers use and it gives an error.

### Are you looking for hardware support?

would it be possible to port over the files needed to allow the lcd for the C2 and H32? 

### Describe the feature you want

I found this github which allows H32 support in Eclipse using the previous Alex's source code for Eclipse IDE. 

https://github.com/smokeypell/Smokey-s_Aquila_C2_H32_Firmware/tree/master

shadow578, 2023-07-31:
hi, since i don't own a printer with such a LCD, i cannot add support for this.

classicrocker883, 2023-08-02:
you could always try to incorporate it and I can have it tested for you.

the voxelab community is very helpful, I can ask to see about donating an LCD for you?

shadow578, 2023-08-05:
i could look into adding support, but it will probably not work right away. 

as a first start: do you know what part of the HAL enables support for this LCD? 
is it handled by the files in [Marlin/src/HAL/STM32F1/dogm/u8g_com_stm32duino_swspi.cpp](https://github.com/shadow578/Marlin-H32/blob/aquila/Marlin/src/HAL/STM32F1/dogm/u8g_com_stm32duino_swspi.cpp)

classicrocker883, 2023-08-09:
yes it is U8GLIB, it needs to load U8GLIB-HAL.h through the Marlin repository.

selecting `CR10_STOCKDISPLAY` instead of a DWIN LCD enables the monochrome screen libraries. there is another repo which was able to have this done.

I believe it is called Smokey-s_Aquila_C2_H32_Firmware

classicrocker883, 2023-08-10:
[HD32_C2-master.zip](https://github.com/shadow578/Marlin-H32/files/12307766/HD32_C2-master.zip)
here is the repo directly from Aquila for the monochrome screen. you should be able to find the required libraries.

shadow578, 2023-08-19:
ok, i'll take a look at it. but it might take a bit of time...

shadow578, 2023-08-29:
at first glance, it looks like these displays just need some sort of SPI library to get them working. Maybe it's even possible to just ~steal~ borrow the one from the STM32 HAL.

---

i'm not quite sure on the pinout. 
from the C2 repo i've found the following pinouts, but i've no idea if they're correct (`RET_12864_LDC` pins).

Pin Name | Pin Assigment
-|-
`LCD_PINS_RS` | `PB12`
`LCD_PINS_ENABLE` | `PB15`
`LCD_PINS_D4` | `PB13`
`BTN_ENC` | `PB1`
`BTN_EN1` | `PC6`
`BTN_EN2` | `PB14`
`BEEPER_PIN` | `PB9`

most of the pins do correlate with the display port found on the `Aquila V1.0.1` /  `Aquila V1.0.2` mainboard i've got. but `PB0` and `PB1` are used for BLTouch on those boards. 
so maybe you've got a different mainboard?

could you please check & confirm the pins

shadow578, 2023-08-29:
see https://github.com/shadow578/Marlin-H32/tree/pio-arduino-core-rewrite_u8glib-test

i've created a quick port of the SPI lib from the STM32 HAL and added some defines. with `CR10_STOCKDISPLAY` enabled in config, it does compile - if it boots i have no idea since i could not test without a matching screen to test with.

as far as pinouts go, i've opted to __not__ set the pinout in `pins_Aquila_V101.h` and instead used placeholders. 
please check & update the pinout before flashing the firmware.

it also seems like the pins in my previous comment are not complete... marlins names are somethimes really kinda cryptic, so no idea if everything needed is defined now...

classicrocker883, 2023-09-14:
I unfortunately do not have the type of LCD to test with. I imagine the pinout should work the same as defined in the pins_CREALITY_V4.h file. perhaps the other repos which actually got it working with that screen can be used as reference.

so far I've been able to get a successful compile, with a couple of warnings however... still no means of testing it, but I can ask the community if someone wishes to volunteer.

this is what I have found:
---

the following warnings in these files showed up

**.pio\libdeps\HC32F460xCxx_AQUILA_V101\U8glib-HAL\src\clib\  
u8g_arduino_fast_parallel.c  
u8g_arduino_no_en_parallel.c  
u8g_com_arduino_t6963.c**

```r
implicit declaration of function 'portOutputRegister'
implicit declaration of function 'portInputRegister'
implicit declaration of function 'portModeRegister'
implicit declaration of function 'digitalPinToBitMask'
```

**Marlin\src\lcd\menu\menu_ubl.cpp**  
```r
'sprintf' may write a terminating nul past the end of the destination
```

I had to change some things in some files inorder to get this to work this far.

in file .platformio\packages\ **framework-arduino-hc32f46x\cores\arduino\delay.h**
from `#include "variant.h"` to  
`#include "../../variants/generic_hc32f460/variant.h"`

in file .platformio\packages\ **framework-arduino-hc32f46x\libraries\Wire\Wire.h**
```y
#include "../../cores/arduino/Stream.h"
#include "../../variants/generic_hc32f460/variant.h"
#include "../../cores/arduino/RingBuffer.h"
```

here is included file changes, including Configuration.h and _adv.h files used to compile.
[hc32-file-changes.zip](https://github.com/shadow578/Marlin-H32/files/12605078/hc32-file-changes.zip)

shadow578, 2023-09-25:
>  I imagine the pinout should work the same as defined in the pins_CREALITY_V4.h file.

you probably mean the one from the voxelab repo, right? I'd like to agree with that, but either i'm missing something (probably tbh) or the pins are incomplete

---

 > the following warnings in these files showed up

i've got those warnings too, it seems like the U8glib library tries to compile part of a HAL for a different platform... probably will be fine...

> edit: in regards to this warning - the more recent Marlin bugfix-2.1.x has been updated since. it is possible some of these errors or warnings could be fixed if using a more recent Marlin source code.

would make sense if it's actually a issue in U8glib

---

> I had to change some things in some files inorder to get this to work this far.

weird, iirc i was able to compile it fine as-is.

> in file .platformio\packages\ framework-arduino-hc32f46x\cores\arduino\delay.h
> from #include "variant.h" to
> #include "../../variants/generic_hc32f460/variant.h"

i'm not quite sure why this would be needed, could you provide more information why this is needed / why you've done this?

> in file .platformio\packages\ framework-arduino-hc32f46x\libraries\Wire\Wire.h

this one's especially weird, because the `Wire` library is not implemented in `framework-arduino-hc32f46x`. So anything that would 'benefit' from changes in that file would have to depend on the `Wire` library, which is not present currently. 
Again, could you provide some more information why this is needed?

classicrocker883, 2023-09-30:
oh you know what, for the Wire.h include, that was something I copied over using a Bootscreen.h file I had, used it for test compiling. I dont recall why the wire.h file had to be included, but i think it worked fine omitting it.

anyway let me re-compile everything and make sure I did it all right.

so when I copied over the framework folders, separately into **.platformio**, im supposed to download those by themselves and then paste manually into there??

because using the other `rewrite`, in the hc32.ini file
```r
[common_HC32_variant]
platform = https://github.com/shadow578/platform-hc32f46x.git
```

this doesnt automatically add the platform frameworks. but wouldn't adding like 
```r
[common_HC32_variant]
platform = https://github.com/shadow578/platform-hc32f46x.git#pio-arduino-core-rewrite_u8glib-test
```
do that for me?? however I dont think the `platform-hc32f46x` is linked directly to the other branch versions for the other framework folders.

shadow578, 2023-10-06:
> oh you know what, for the Wire.h include, that was something I copied over using a Bootscreen.h file I had, used it for test compiling. I dont recall why the wire.h file had to be included, but i think it worked fine omitting it.

that would make sense

---

regarding the branches, the way i'm switching between branches is by
1. letting pio install everything normally
2. removing the stuff pio installed
3. manually `git clone`-ing the framework and platform repos
4. copying `.git/.piopm` from the folders pio installed into the manually cloned repos

it's a bit messy but afaik platformio does not have a way to specify a specific branch...

classicrocker883, 2023-10-06:
> it's a bit messy but afaik platformio does not have a way to specify a specific branch...

*this should* work, add `#[branch]` to the end of **.git** like this
```r
[common_HC32_variant]
platform = https://github.com/shadow578/platform-hc32f46x.git#arduino-core-rewrite
```

but for the file such as in the json  
it *should* work here too
```json
"version": "https://github.com/shadow578/framework-arduino-hc32f46x.git#arduino-core-rewrite"
```

classicrocker883, 2023-10-07:
tried compiling for 12864 LCD (`CR10_STOCKDISPLAY`)

in **pins_Aquila_V101.h**  
I added below `// Screen Pins`
```y
#if ENABLED(CR10_STOCKDISPLAY)
#undef LCD_SERIAL_PORT
#define LCD_SERIAL_PORT 1

  #define LCD_PINS_RS                EXP_07_PIN
  #define LCD_PINS_EN                EXP_08_PIN
  #define LCD_PINS_D4                EXP_06_PIN

  #define BTN_ENC                    EXP_02_PIN
  #define BTN_EN1                    EXP_03_PIN
  #define BTN_EN2                    EXP_05_PIN

  #ifndef HAS_PIN_27_BOARD
    #define BEEPER_PIN               EXP_01_PIN
  #endif

#elif ANY(HAS_DWIN_E3V2, IS_DWIN_MARLINUI)
```

and everything still compiles successful and fine, except for this:
```r
.pio\libdeps\HC32F460xCxx_AQUILA_V101\U8glib-HAL\src\clib\u8g_com_arduino_fast_parallel.c:
In function 'u8g_com_arduino_fast_parallel_init':
warning: implicit declaration of function 'portOutputRegister' + 'digitalPinToPort' [-Wimplicit-function-declaration]
  103 |   u8g_data_port[0] = (u8g_ptr_t)portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
warning: implicit declaration of function 'digitalPinToBitMask' [-Wimplicit-function-declaration]
  104 |   u8g_data_mask[0] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D0]);

.pio\libdeps\HC32F460xCxx_AQUILA_V101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:
In function 'u8g_com_arduino_no_en_parallel_init':
warning: implicit declaration of functions 'digitalPinToPort' + 'portOutputRegister' [-Wimplicit-function-declaration]
   88 |   u8g_data_port[0] = (u8g_ptr_t)portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
warning: implicit declaration of function 'digitalPinToBitMask' [-Wimplicit-function-declaration]
   89 |   u8g_data_mask[0] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D0]);

.pio\libdeps\HC32F460xCxx_AQUILA_V101\U8glib-HAL\src\clib\u8g_com_arduino_t6963.c:
In function 'u8g_com_arduino_t6963_init':
warning: implicit declaration of function 'portOutputRegister' [-Wimplicit-function-declaration]
   94 |   u8g_output_data_port[0] = (u8g_ptr_t)portOutputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
warning: implicit declaration of function 'portInputRegister' [-Wimplicit-function-declaration]
   95 |   u8g_input_data_port[0] = (u8g_ptr_t)portInputRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));
warning: implicit declaration of function 'portModeRegister' [-Wimplicit-function-declaration]
   96 |   u8g_mode_port[0] = (u8g_ptr_t)portModeRegister(digitalPinToPort(u8g->pin_list[U8G_PI_D0]));  
warning: implicit declaration of function 'digitalPinToBitMask' [-Wimplicit-function-declaration]
   97 |   u8g_data_mask[0] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D0]);
```

classicrocker883, 2023-10-10:
the above functions are not defined, and should be in one of the framework files.  
for other framework packages, this is how they have it:

**packages\framework-arduinoststm32\cores\arduino\pins_arduino.h**
```y
#define digitalPinToPort(p)         (get_GPIO_Port(STM_PORT(digitalPinToPinName(p))))
#define digitalPinToBitMask(p)      (STM_GPIO_PIN(digitalPinToPinName(p)))

#define analogInPinToBit(p)         (STM_GPIO_PIN(digitalPinToPinName(p)))
#define portOutputRegister(P)       (&(P->ODR))
#define portInputRegister(P)        (&(P->IDR))
```

---
**packages\framework-arduino-avr\cores\arduino\Arduino.h**  
```y
#define digitalPinToPort(P) ( pgm_read_byte( digital_pin_to_port_PGM + (P) ) )
#define digitalPinToBitMask(P) ( pgm_read_byte( digital_pin_to_bit_mask_PGM + (P) ) )
...
#define portOutputRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_output_PGM + (P))) )
#define portInputRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_input_PGM + (P))) )
#define portModeRegister(P) ( (volatile uint8_t *)( pgm_read_word( port_to_mode_PGM + (P))) )

```

---
**buildroot\share\PlatformIO\variants\marlin_maple_MEEB_3DP\variant.h**
```y
#define digitalPinToPort(P)        ( PIN_MAP[P].gpio_device )
#define digitalPinToBitMask(P)     ( BIT(PIN_MAP[P].gpio_bit) )
#define portOutputRegister(port)   ( &(port->regs->ODR) )
#define portInputRegister(port)    ( &(port->regs->IDR) )
```

what I suggest is we copy -> tweak this code so those functions can be defined to mitigate the warnings, and hopefully this solves being able to use the 128x64 monochrome LCD.

shadow578, 2023-10-13:
(continuation of notes above)

> portOutputRegister wouldn't be possible tho, since it seems like setting (HIGH) and resetting (LOW) uses different registers:
>
>    set: POSRx = POSRA + (0x10 * port) (cf `PORT_SetBits)
>    reset: PORRx = PORRA  + (0x10 * port) (cf PORT_ResetBits)
>
>tho it'd be weird if the GPIO peripheral is that different from the STM32, seeing how the HC32 already copies so much else from it...

seems like the GPIO peripheral has two modes of operation: 
1. using three registers for clear (`PORRx`), set (`POSRx`) and flip (`POTRx`)
2. using a output data register `PODRx` (cf 9.3.1, section 'general purpose output function', p 288)

interestingly, the DDL basically never uses the second option, even in examples...

but assuming that it works as described (there's no errata entry for it yet, so chances are good), then `portOutputRegister` would have to return `PODRx = PODRA + (0x10 * port)`

---

`portModeRegister` usage:
> For an output, you'll clear the bit, and for an input, you'll set the bit.

the DDL uses the `POUTE` bit in the pin-specific `PCRxy` register, but it seems like using `POERx` has the same function (cf 9.4.11, p 303).
`POERx` is *almost* the right thing, with three caveats:
1. as a prerequisite, at a minimum the `DDIS` in `PCRxy` has to be reset, tho this may be the default state (9.4.11, p 303)
2. the register behaves inversely, so a `0` disables output and a `1` enables output (9.4.3, p 298)
3. the manual never actually mentions if the input function is enabled when set to `0`, it only states that 'the output function is disabled'

shadow578, 2023-10-13:
> `portModeRegister` usage:
> >   For an output, you'll clear the bit, and for an input, you'll set the bit.

turns out that was just plain wrong... on AVR, `portModeRegister` would return `DDRB` and the likes.
`DDRB` sets the pin as input when set to `0`, and as output when set to `1`.

that would leave `portModeRegister` as `POERx = POERA + (0x10 * port)`. 

---

with that, every macro ug8lib complained about would be available, at least theoretically.

classicrocker883, 2023-10-14:
> on another note, in what context are these functions used? maybe they're used in code that isn't actually needed since the HAL comes with it's own driver (?)
> 
> or maybe there is a setting to force the lib to use 'normal' arduino functions (`digitalWrite` etc...)

that may be it...maybe these functions arent needed. so others have used Eclipse to compile for the H32 using this type of LCD screen right? Is all the code and libraries they used the same when porting over to be used in PlatformIO? I mean more specifically for this LCD in general. I did a quick compare of the files they used and everything seems about the same, except maybe for one or two. But I dont recall seeing something pertaining to like these functions, but it might be good to take another look now knowing what to look for.

anyway so everything compiles okay except for just these couple of warnings, I mean they arent errors and it ends in being successful, unfortunately I do not have an LCD screen to test myself, I will ask around the forums maybe someone can test it out as is. how does that sound?

shadow578, 2023-10-15:
> that may be it...maybe these functions arent needed. so others have used Eclipse to compile for the H32 using this type of LCD screen right? Is all the code and libraries they used the same when [...]

you know what, i think i know why it's only here where these issues are popping up: every other Marlin port for teh HC32 sticks fairly close to the source published by Voxelab, Creality, ...; thus needing to compile with marlin. In those version, the HC32 HAL is just kinda 'slapped' over the STM32 HAL, predending like the HC32 is a STM32 (including setting macros that would identify the build as for a STM32, so `ARDUINO_ARCH_STM32` and the likes).

but since i've completely re-written the HAL and everything, i'm not setting the STM32 flags anymore. So theres probably some checks that disable these functions when compiling for STM32...

---

> anyway so everything compiles okay except for just these couple of warnings, I mean they arent errors and it ends in being successful, unfortunately I do not have an LCD screen to test myself, I will ask around the forums maybe someone can test it out as is. how does that sound?

would be great to find someone that has a LCD to test. maybe point them to the issue too, so they can give feedback more directly.

shadow578, 2023-10-15:
> with that, every macro ug8lib complained about would be available, at least theoretically.

i've now added all the macros, and testing them worked perfectly as expected (even almost on the first try :))
see https://github.com/shadow578/framework-arduino-hc32f46x/pull/5

with that, compiling no longer shows the warnings. i'm wondering if this means that everything will just work(™) now, or if theres more things missing

classicrocker883, 2023-10-20:
awesome work! very nice. I'll reply back soon as a test is done

classicrocker883, 2023-10-22:
so far it does compile fine without any warnings, very nice job

one thing to note, at first compile I got this warning

```r
Marlin\src\lcd\menu\menu_ubl.cpp: In function 'void _lcd_ubl_adjust_height_cmd()':
Marlin\src\lcd\menu\menu_ubl.cpp:150:39: warning: 'sprintf' may write a terminating nul past the end of the destination [-Wformat-overflow=]
  150 |   sprintf_P(&ubl_lcd_gcode[ind], PSTR(".%i"), ABS(ubl_height_amount));
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/avr/pgmspace.h:99:44: note: in definition of macro 'sprintf_P'
   99 | #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
      |                                            ^
Marlin\src\lcd\menu\menu_ubl.cpp:150:34: note: in expansion of macro 'PSTR'     
  150 |   sprintf_P(&ubl_lcd_gcode[ind], PSTR(".%i"), ABS(ubl_height_amount));
      |                                  ^~~~
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/avr/pgmspace.h:99:37: note: 'sprintf' output between 3 and 8 bytes into a destination of size 7
   99 | #define sprintf_P(s, f, ...) sprintf((s), (f), __VA_ARGS__)
      |                              ~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~
Marlin\src\lcd\menu\menu_ubl.cpp:150:3: note: in expansion of macro 'sprintf_P' 
  150 |   sprintf_P(&ubl_lcd_gcode[ind], PSTR(".%i"), ABS(ubl_height_amount));  
```

but after putting in a new configuration.h + _adv.h file it compiled fine without the warning.

figure I would just mention it anyway just incase. eventhough it does compile normally.

otherwise disregard the warning, so nothing to worry about now. 

just need someone to test it out with that type of LCD...someday soon!

shadow578, 2023-10-25:
the warning isn't related to HC32, other than HC32 using a more recent version of gcc that catches this bug. 
you've actually just found a (potential) bug in Marlin.

---

as for the bug itself:

1. `ubl_lcd_gcode` is initialized with a length of 13, so it can effectively hold a string of 12 characters (the last "slot" is needed for a terminating `\0`)
2. the string "G29P6C-" is copied into `ubl_lcd_gcode`, filling the first 7 "slots" of the array
3. `sprintf` formats `ubl_height_amount` with `.%i`, so `ubl_height_amount=123` would be formatted as `.123`
	- if `ubl_height_amount` is positive, the last char '-' from 2. is overwritten

the issue is that if `ubl_height_amount` ever is larger than 99999 (or smaller than -9999), `sprintf` will write data past the end of `ubl_lcd_gcode`.

the bug could be mitigated by either limiting `ubl_height_amount` to the "safe" range, or alternatively, replacing `sprintf` with `snprintf`.

smokeypell, 2024-01-01:
Shadow578 I would be willing to help in any way that I can. I'm not much of a developer, but I can try to provide any testing and as much feedback as I can. The only real contribution to this cause I have made is [Smokey's_Aquila_C2_H32_Firmware](https://github.com/smokeypell/Smokey-s_Aquila_C2_H32_Firmware). It was a pain, but nothing really groundbreaking. 
I was basically just trying to get something out there that offered more than just basic garbage functionality (Plus making sure that Thermal Runaway was enabled so everyone could have safer printers). I would like to try to help get all this figured out and squared away so that we can get these boards and chips integrated into mainstream. I find it a bit of a pain to have so many machines without mainstream Marlin support. If we could get this figured out, it would benefit tons of people with Voxelab machines as well as Anycubic machines running these chips.

shadow578, 2024-01-02:
hi @smokeypell 

first off, thank you for you offer to help. 

> we can get these boards and chips integrated into mainstream

This has already happened. Please checkout the `bugfix-2.1.x` branch on either this repo or the MarlinFirmware/Marlin repo. 
Both should contain a directory `Marlin/src/HAL/HC32/`. 

There's even a (initial / experimental) support for the C2's screen included. 
Tho you may have to double-check the pin definitions (see [pins_AQUILA_101.h#L155](https://github.com/MarlinFirmware/Marlin/blob/5b74e25108a47acad41d9a50560cd1fbae38040a/Marlin/src/pins/hc32f4/pins_AQUILA_101.h#L155)). From a first glance, it seems to match up with the pins for [RET6 12864 LCD](https://github.com/Voxelab-64/HD32_C2/blob/b533ea06e262536bd51d2933fc49980a714fdc36/source/Marlin/src/pins/stm32f1/pins_CREALITY_V427.h#L169) in voxelab's C2 repo.

So if you want, you can go ahead and compile Marlin and test if the current implementation does work. 
A few things to look out for:
1. you'll have to transfer the values in `Configuration.h` and `Configuration_adv.h`. In this repo's issues there is one containing sample configurations for the X2, so start from those and adjust the config where needed.
2. double-check the pin definitions for the LCD, as they may be wrong currently. If you feel comfortable doing so, you could also check the physical wiring using a multimeter. Let me know if you'd like to go that route, let me know so i'll provide pinouts for the HC32.

if you encounter any issues (screen not working, random crashes / reboots), please let me know in this issue

smokeypell, 2024-01-03:
@shadow578 I tried to compile Marlin bugfix-2.1.x and your bugfix-2.1.x branches with no luck. Everything compiles fine but no luck on the screen. I added a new pin file for the C2 board. It is a different board with a different pinout. Different pins for the screen and some other things. This was something I confirmed with Voxelab last year when I was working on compiling their HD32_C2 repository code.

I wish I could help dig into the HAL issue, as I suspect that's the entirety of the issue. Unfortunately, I'm not that skilled.

I forked your bugfix-2.1.x repository and added the files and definitions mentioned. I haven't requested a pull request. I figured I'd let you check it out first. [https://github.com/smokeypell/Marlin-H32/tree/bugfix-2.1.x](https://github.com/smokeypell/Marlin-H32/tree/bugfix-2.1.x)

closed 2024-02-29 by shadow578

classicrocker883, 2024-07-26:
I would like to revisit this but using a different 12864 LCD.
if you need the config files I used let me know, but these are the changes I had to make to get it to work, so far I've been only to compile successfully, next step is flashing firmware and connecting the LCD. However, the pinout's aren't plug and play, I can try to work with however it is defined in **pins_CREALITY_V4.h** (which I copied over to HC32 pins file), except there are some things I am not sure about, mainly the RGB backlight stuff; if `NEOPIXEL_PIN` is the right one.

Also, the actually defined pinouts, from board to LCD EXP1 + EXP2. this requires custom 2 cables to be merged into one. for now I'm experimenting with figuring out what is the best course of action regarding how to *redefine* each board pinout using https://github.com/MarlinFirmware/Marlin/issues/18225 as a guide

in **Configuration.h**:
```y
#define MKS_MINI_12864_V3
#define NEOPIXEL_LED
#define NEOPIXEL_TYPE          NEO_RGB 
#define NEOPIXEL_PIN               PA13
```

**Marlin\src\HAL\HC32\inc\Conditionals_LCD.h** doesn't actually do anything:
```y
#if ALL(HAS_MARLINUI_U8GLIB, FORCE_SOFT_SPI)
  #define U8G_SW_SPI_HC32 1
#endif
```

---
changes made (using Marlin bugfix-2.1.x)
<details>
<summary>file: <b>Marlin/src/HAL/HC32/u8g/u8g_com_HAL_HC32_sw_spi.cpp</b></summary>

```diff
diff --git a/Marlin/src/HAL/HC32/u8g/u8g_com_HAL_HC32_sw_spi.cpp b/Marlin/src/HAL/HC32/u8g/u8g_com_HAL_HC32_sw_spi.cpp
index 97f4b85eaf..d081a618e4 100644
--- a/Marlin/src/HAL/HC32/u8g/u8g_com_HAL_HC32_sw_spi.cpp
+++ b/Marlin/src/HAL/HC32/u8g/u8g_com_HAL_HC32_sw_spi.cpp
@@ -24,7 +24,7 @@
 
 #include "../../../inc/MarlinConfig.h"
 
-#if U8G_SW_SPI_HC32
+#if ALL(HAS_MARLINUI_U8GLIB, FORCE_SOFT_SPI)
 
 #warning "Software SPI for U8Glib is experimental on HC32F460. Please share your experiences at https://github.com/shadow578/Marlin-H32/issues/35"
 
@@ -39,7 +39,7 @@
 static uint8_t SPI_speed = LCD_SPI_SPEED;
 
 static inline uint8_t swSpiTransfer_mode_0(uint8_t b, const uint8_t spi_speed, const pin_t miso_pin=-1) {
-  for (i = 0; i < 8; ++i) {
+  for (uint8_t i = 0; i < 8; ++i) {
     if (spi_speed == 0) {
       WRITE(DOGLCD_MOSI, !!(b & 0x80));
       WRITE(DOGLCD_SCK, HIGH);
@@ -49,20 +49,20 @@ static inline uint8_t swSpiTransfer_mode_0(uint8_t b, const uint8_t spi_speed, c
     }
     else {
       const uint8_t state = (b & 0x80) ? HIGH : LOW;
-      for (j = 0; j < spi_speed; ++j) WRITE(DOGLCD_MOSI, state);
-      for (j = 0; j < spi_speed + (miso_pin >= 0 ? 0 : 1; ++j)) WRITE(DOGLCD_SCK, HIGH);
+      for (uint8_t j = 0; j < spi_speed; ++j) WRITE(DOGLCD_MOSI, state);
+      for (uint8_t j = 0; j < spi_speed + (miso_pin >= 0 ? 0 : 1); ++j) WRITE(DOGLCD_SCK, HIGH);
 
       b <<= 1;
       if (miso_pin >= 0 && READ(miso_pin)) b |= 1;
 
-      for (j = 0; j < spi_speed; ++j) WRITE(DOGLCD_SCK, LOW);
+      for (uint8_t j = 0; j < spi_speed; ++j) WRITE(DOGLCD_SCK, LOW);
     }
   }
   return b;
 }
 
 static inline uint8_t swSpiTransfer_mode_3(uint8_t b, const uint8_t spi_speed, const pin_t miso_pin=-1) {
-  for (i = 0; i < 8; ++i) {
+  for (uint8_t i = 0; i < 8; ++i) {
[... 62 lines cut ...]
+   * Needs custom cable. Connect EN2-EN2, LCD_CS-LCD_CS and so on.
+   * Debug port is just above EXP1. You need to add pins.
+   *
+   */
+
+  #define BTN_ENC                    EXP1_02_PIN // PB2
+  #define BTN_EN1                    EXP1_01_PIN // PC6
+  #define BTN_EN2                    EXP1_06_PIN // PB13
+  #define BEEPER_PIN                        -1
+
+  #define DOGLCD_CS                  EXP1_03_PIN // PB10
+  #define DOGLCD_A0                  EXP1_05_PIN // PB14
+  #define DOGLCD_SCK                 EXP1_07_PIN // PB12
+  #define DOGLCD_MOSI                EXP1_08_PIN // PB15
+  #define LCD_RESET_PIN              EXP1_04_PIN // PB11
+
+  #define FORCE_SOFT_SPI
+  #define LCD_BACKLIGHT_PIN                 -1
+  #ifndef NEOPIXEL_LED
+    #define NEOPIXEL_PIN                    PA13
+  #endif
+
 #endif
```
</details>