shadow578/Marlin-H32 #45: [BUG] Some terms should be updated

classicrocker883, 2023-10-07:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

for instance:

in **src/HAL/HC32F46x/inc/Conditionals_post.h**
``` diff
#if USE_FALLBACK_EEPROM
#define SDCARD_EEPROM_EMULATION
-#elif EITHER(I2C_EEPROM, SPI_EEPROM)
+#elif ANY(I2C_EEPROM, SPI_EEPROM)
#define USE_SHARED_EEPROM 1
#endif

// Allow SDSUPPORT to be disabled
-#if DISABLED(SDSUPPORT)
-#undef SDIO_SUPPORT
+#if !HAS_MEDIA
+#undef ONBOARD_SDIO
#endif

```

### Version of Marlin Firmware

current

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-10-07:
this may be a bit off topic, but I thought I would share this if you dont know about it already

https://www.xhsc.com.cn/Productlist/info.aspx?itemid=1853&parent

if you convert it to english, under the development tools tab you will see something like this.  
updated files for this chip
[HC32F460_DDL_Rev3.1.0.zip](https://github.com/shadow578/Marlin-H32/files/12836684/HC32F460_DDL_Rev3.1.0.zip)

shadow578, 2023-10-07:
hi, thanks for opening the issue. i've merged the upstream changes yesterday and forgot to apply some changes to the HAL before merging, so rn the `aquila` branch is broken. i'm working on a fix in https://github.com/shadow578/Marlin-H32/pull/44 already.

---

> this may be a bit off topic, but I thought I would share this if you dont know about it already
>
> https://www.xhsc.com.cn/Productlist/info.aspx?itemid=1853&parent
>
> if you convert it to english, under [...]

i'm aware of that page, but i did not yet notice that the DDL has been updated. 
i'll look into the changes and update the ddl framework sometime then

closed 2023-10-07 by shadow578