shadow578/Marlin-H32 #51: [BUG] Failed to build, compile firmware using Actions

classicrocker883, 2023-10-26:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

this I believe has more to do with platform and framework repo.

but I was trying to do a github action to build the firmware and it has always worked before, now I get this error

```r
Tool Manager: framework-hc32f46x-ddl@2.2.0+sha.e249973 has been installed!
Verbose mode can be enabled via `-v, --verbose` option
/bin/sh: 1: Syntax error: "(" unexpected
Error: Failed to parse Marlin features. See previous error messages.
========================= [FAILED] Took 25.09 seconds =========================

Environment               Status    Duration
------------------------  --------  ------------
HC32F4[60](https://github.com/classicrocker883/MRiscoCProUI/actions/runs/6648180376/job/18064816887#step:8:61)xCxx_AQUILA_V101  FAILED    00:00:25.086
==================== 1 failed, 0 succeeded in 00:00:25.086 ====================
Error: Process completed with exit code 1.
```

not sure where or what it means by syntax error. but apparently there is an extra "(" or something.

it compiles fine using VSCode, however not using github actions. not sure why or what changed.

### Version of Marlin Firmware

recent version

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-10-26:
it must have something to do with **src/HAL/HC32F46X** because I tried using a previous older branch and it got farther in the build, it at least started to compile where as this recent one did not.

but 3 weeks ago or so I was able to build using actions fine. so somewhere in that timeframe, i believe it could be inside the HAL, but also it could be in a **framework-arduino/ddl** or platform....

classicrocker883, 2023-10-26:
hey were getting somewhere!

so as I suspected, it was in the **hc32.ini** file... I changed some things...

then got this error
```r
Compiling .pio/build/HC32F460xCxx_AQUILA_V101/FrameworkArduino/CORE/IPAddress.cpp.o
In file included from /home/runner/.platformio/packages/framework-arduino-hc32f46x/cores/arduino/IPAddress.cpp:22:
/home/runner/.platformio/packages/framework-arduino-hc32f46x/cores/arduino/Arduino.h:73:10: fatal error: drivers/usart/Usart.h: No such file or directory
   73 | #include "drivers/usart/Usart.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

what I did was comment out 
```r
#-D __CORE_DEBUG
#-D __DEBUG_SHORT_FILENAMES
#-D __CORE_DEBUG_SHORT_FILENAMES
#-D __CORE_DEBUG_OMIT_PANIC_MESSAGE
```

but most importantly, I removed the parentheses here `-D F_CPU=SYSTEM_CLOCK_FREQUENCIES.pclk1`

ill keep trying to see what works.

classicrocker883, 2023-10-28:
the odd thing is the workflow worked normally before, despite the line in question causing the error now has been there the whole time.

[8_Run PlatformIO.txt](https://github.com/shadow578/Marlin-H32/files/13195953/8_Run.PlatformIO.txt)  
this .txt is where the error shows.

[0_Build BL-Touch ProUI Files (HC32).txt](https://github.com/shadow578/Marlin-H32/files/13195965/0_Build.BL-Touch.ProUI.Files.HC32.txt)  
this also like the same as first, error shows up near end of file.

[logs_6384.zip](https://github.com/shadow578/Marlin-H32/files/13195954/logs_6384.zip)  
I just pulled two of the .txt files showing the error. here is the full log download if you wish to see the whole workflow.

shadow578, 2023-10-29:
you've mentioned that the workflow worked normally before. Do you happen to have logs of a previous (successfull) run?

shadow578, 2023-10-29:
i've started investigating now. 
All issues here are reproducible __only__ on linux, windows works fine.

1. `/bin/sh: 1: Syntax error: "(" unexpected`

this seems to be caused because platformio runs all compiler commands in the `sh` command interpreter, which *for whatever reason* tries to interpret it as more than just a literal string.

just removing the parentheses as you've suggested seems to work without any issues.

2. `fatal error: drivers/usart/Usart.h: No such file or directory`

this seems to be caused because on linux, filenames are case-sensitive, while on windows they are not. 
And for *some* reason, the header `drivers/usart/Usart.h` is spelled `drivers/usart/usart.h` (lowercase 'U') on github.

this is really weird to me, because i've always spelled it with a uppercase 'U', and even in my local repo it's spelled that way...
so really weird issue, but i'll just rename the file to be spelled correctly and that should be it.

3. extra include

> also, inside framework-arduino-hc32f46x/cores/arduino/drivers/usart/usart.h
> #include "drivers\usart\usart.h" can be omitted?? because it is including itself which is a deadend ya know?

oops, i'll remove it 

---

with these changes, i've sucessfully build the firmware on [WSL](https://learn.microsoft.com/de-de/windows/wsl/about), so there shouldn't be any issues building in a github action workflow.

closed 2023-10-29 by shadow578