shadow578/Marlin-H32 #56: [BUG] CR10_STOCKDISPLAY (12864) support now fails.

classicrocker883, 2023-11-27:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

Remember when having monochrome 12864 LCD was supported, I think the same kind of error is back.

```r
.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t   
```
sorry for the long list, but basically it shows the same message...
`struct "pin_info_t" has no field "bit_mask"`

this is within **.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c**
```y
static void u8g_com_arduino_no_en_parallel_init(u8g_t *u8g)
{ ...
  u8g_data_mask[0] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D0]);
|             #1-#6                                                      #1-#6   |
...
  u8g_data_mask[7] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D7]);
}
```

```r
In file included from C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/Arduino.h:115,
                 from .pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:69:
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c: In function 'u8g_com_arduino_no_en_parallel_init':
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t     
      |                            ^
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:89:34: note: in expansion of macro 'digitalPinToBitMask'     
   89 |   u8g_data_mask[0] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D0]);
      |                                  ^~~~~~~~~~~~~~~~~~~
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t     
      |                            ^
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:91:34: note: in expansion of macro 'digitalPinToBitMask'     
   91 |   u8g_data_mask[1] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D1]);
      |                                  ^~~~~~~~~~~~~~~~~~~
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t     
      |                            ^
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:93:34: note: in expansion of macro 'digitalPinToBitMask'     
   93 |   u8g_data_mask[2] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D2]);
      |                                  ^~~~~~~~~~~~~~~~~~~
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t     
      |                            ^
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:95:34: note: in expansion of macro 'digitalPinToBitMask'     
   95 |   u8g_data_mask[3] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D3]);
      |                                  ^~~~~~~~~~~~~~~~~~~
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t     
      |                            ^
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:98:34: note: in expansion of macro 'digitalPinToBitMask'     
   98 |   u8g_data_mask[4] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D4]);
      |                                  ^~~~~~~~~~~~~~~~~~~
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t     
      |                            ^
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:100:34: note: in expansion of macro 'digitalPinToBitMask'    
  100 |   u8g_data_mask[5] = (u8g_data_t)digitalPinToBitMask(u8g->pin_list[U8G_PI_D5]);
      |                                  ^~~~~~~~~~~~~~~~~~~
C:\Users\Andrew\.platformio\packages\framework-arduino-hc32f46x\cores\arduino/drivers/gpio/direct_access.h:28:28: error: 'pin_info_t' has no member named 'bit_mask'
   28 |     ((uint16_t)PIN_MAP[pin].bit_mask()) // (uint16_t)en_pin_t     
      |                            ^
.pio\libdeps\HC32F460C_aquila_101\U8glib-HAL\src\clib\u8g_com_arduino_no_en_parallel.c:102:34: note: in expansion of macro 'digitalPinToBitMask'    
[... 61 lines cut ...]
```

### Bug Timeline

since recent update

### Version of Marlin Firmware

current up to date

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

classicrocker883, 2023-11-27:
I took a quick look back (in time) at a previous commit [optimize PIN_MAP array](https://github.com/shadow578/framework-arduino-hc32f46x/commit/bd8172d) from your _framework-arduino-hc32f46x_

I changed:
```diff
#define digitalPinToBitMask(pin) \
-    ((uint16_t)PIN_MAP[pin].bit_mask())
#define digitalPinToBitMask(pin) \
+    ((uint16_t)PIN_MAP[pin].bit_mask)
```
from **\.platformio\packages\framework-arduino-hc32f46x\cores\arduino\drivers\gpio\direct_access.h**

and  
```diff
#ifdef __cplusplus
		/**
		 * @brief bit mask of the pin in the port
		 */
		en_pin_t bit_mask() const
		{
			return (en_pin_t)(1 << bit_pos);
		}
+#else

+		en_pin_t bit_mask;

#endif
```
from _\.platformio\packages\framework-arduino-hc32f46x\cores\arduino\WVariant.h_

I got a successful compile from this with no other warnings...
however, I do not feel this is the actual correct way it should be permanently. I mean I haven't test if it will even work.

but maybe its a step in the right direction?

shadow578, 2023-12-04:
fixed by [222eb2b](https://github.com/shadow578/framework-arduino-hc32f46x/commit/222eb2b2ae5e4de78f11933326ba9d3a9184f989) in arduino repo

closed 2023-12-04 by shadow578