shadow578/Marlin-H32 #61: [BUG] LCD Display just blank with backlight on, on Ender 2 Pro

jam-berg, 2024-05-09:
### Did you test the latest `bugfix-2.1.x` code?

Yes, and the problem still exists.

### Bug Description

The Build works fine for the HC32F460 Chip in Upstream with the corresponding board for the Ender 2 Pro. Serial Communication also works as far as I was able to test it.
However after flashing the screen is just blank. Reflashing to the Stock firmware makes it work again. 
The Serial communication also prints out the correct build I built from Marlin.

I also noticed that the M100 - Free Memory gcode command is not implemented.
Is the LCD Screen already implemented for the HC32F460?
I tried multiple configurations, including the default config from marlin for my board and the official from creality in their repository for this CPU https://github.com/CrealityOfficial/Ender-2-Pro/tree/MCU_HDSC 

Is there anything further I can test or help with?

### Bug Timeline

new since HC32F460 support was recently added

### Expected behavior

I expect, that after starting the printer, the LCD shows the splash screen and then shows the infoscreen.

### Actual behavior

It starts, turns the LCD Backlight on, but no image.

### Steps to Reproduce

1. Change Config of Motherboard to BOARD_CREALITY_ENDER2P_V24S4
2. Compile
3. Flash with binary in "Factory" Folder with unique name
4. Power on printer an wait for several minutes
5. reboot when SD Card is removed

### Version of Marlin Firmware

bugfix-2.1.x

### Printer model

Ender 2 Pro

### Electronics

BOARD_CREALITY_ENDER2P_V24S4, HC32F460KCTA

### LCD/Controller

stock, probably CR-10

### Bed Leveling

None

### Your Slicer

None

### Host Software

None

### Don't forget to include

- [X] A ZIP file containing your `Configuration.h` and `Configuration_adv.h`.

jam-berg, 2024-05-09:
[image]
This is an image of the board, if it helps.

jam-berg, 2024-05-09:
Managed to solve it. I changed the content of the hc32f4/pins_CREALITY_ENDER2P_V24S4.h file to the following:

`

/**
 * CREALITY v4.2.7 (STM32F103) board pin assignments
 */
#define RET6_12864_LCD
#if defined(CR10_STOCKDISPLAY) && !defined RET6_12864_LCD &&                   \
    !definedVET6_12864_LCD
#error                                                                         \
    "Define RET6_12864_LCD or VET6_12864_LCD to select pins for CR10_STOCKDISPLAY with the Creality V4 controller."
#endif

#if HOTENDS > 1 || E_STEPPERS > 1
#error                                                                         \
    "Creality V4 only supports one hotend / E-stepper. Comment out this line to continue."
#endif

#define BOARD_INFO_NAME "Creality v2.4.S4"
#define DEFAULT_MACHINE_NAME "Creality3D"

#define BOARD_NO_NATIVE_USB
/* LED0 Port/Pin definition */
#define LED0 PC3 // TEST
/*
 * SPI3 Pins
 */
#define SPI_NSS PB15  // SPI3_NSS
#define SPI_SCK PB14  // SPI3_CLK
#define SPI_MISO PB13 // SPI3_MISO
#define SPI_MOSI PB12 // SPI3_MOSI

/*
 * SDIO Pins
 */
#define BOARD_SDIO_D0 PC8
#define BOARD_SDIO_D1 PC9
#define BOARD_SDIO_D2 PC10
#define BOARD_SDIO_D3 PC11
#define BOARD_SDIO_CLK PC12
#define BOARD_SDIO_CMD PD2
#define BOARD_SDIO_DET PA10

// USARTS

#define BOARD_USART2_TX_PIN PA8 // USB
#define BOARD_USART2_RX_PIN PA7

#define BOARD_USART1_TX_PIN PA2 // LCD
#define BOARD_USART1_RX_PIN PA3

// Onboard crystal oscillator
//
#ifndef BOARD_XTAL_FREQUENCY
#define BOARD_XTAL_FREQUENCY 8000000 // 8 MHz XTAL
#endif

//
// EEPROM
//
#define IIC_BL24CXX_EEPROM // EEPROM on I2C-0
// #define SDCARD_EEPROM_EMULATION

#ifdef IIC_BL24CXX_EEPROM
#define IIC_EEPROM_SDA PA12
#define IIC_EEPROM_SCL PA11
#define MARLIN_EEPROM_SIZE 0x800 // 2Kb (24C16)
#endif

//
// Servos
//
#ifndef SERVO0_PIN
#define SERVO0_PIN PB2 // BLTouch OUT PIN
#endif

//
// Limit Switches
//
#define X_STOP_PIN PA5
#define Y_STOP_PIN PA6
#define Z_STOP_PIN PB0
#define X_MAX_PIN PA4
#define Y_MAX_PIN -1
#define Z_MAX_PIN -1

#ifndef Z_MIN_PROBE_PIN
#define Z_MIN_PROBE_PIN PB1 // BLTouch IN
#endif

//
// Filament Runout Sensor
//
#ifndef FIL_RUNOUT_PIN
#define FIL_RUNOUT_PIN PC15 // "Pulled-high"
#endif

//
// Steppers
//
#define X_ENABLE_PIN PC2
#define X_STEP_PIN PC1
#define X_DIR_PIN PC0

#define Y_ENABLE_PIN PH2
#define Y_STEP_PIN PB9
#define Y_DIR_PIN PB8

#define Z_ENABLE_PIN PB7
#define Z_STEP_PIN PB6
#define Z_DIR_PIN PB5

#define E0_ENABLE_PIN PB4
#define E0_STEP_PIN PB3
#define E0_DIR_PIN PA15

//
// Temperature Sensors
//
#define TEMP_0_PIN PC5   // HEATER1 ADC1_IN15
#define TEMP_BED_PIN PC4 // HOT BED ADC1_IN14
#define ADC_CHANNEL_COUNT 2u

//
// Heaters / Fans
//
#define HEATER_0_PIN PA1    // HEATER1
#define HEATER_BED_PIN PB10 // HOT BED

#ifndef FAN0_PIN
#define FAN0_PIN PA0 // FAN
#endif
#if PIN_EXISTS(FAN)
#define FAN_SOFT_PWM
#endif

//
// SD Card
//
#define SD_DETECT_PIN PA10
#define SDCARD_CONNECTION ONBOARD
#define ONBOARD_SPI_DEVICE 1
#define ONBOARD_SD_CS_PIN PC3 // SDSS
#define ONBOARD_SDIO
#define NO_SD_HOST_DRIVE // This board's SD is only seen by the printer

#if ENABLED(RET6_12864_LCD)

/* RET6 12864 LCD */
#define LCD_PINS_RS PB15 // SPI3_NSS
#define LCD_PINS_EN PB12 // SPI3_MOSI
#define LCD_PINS_D4 PB14 // SPI3_CLK

#define BTN_ENC PC7  // LCD_RST
#define BTN_EN1 PA2  // UART4-TX
#define BTN_EN2 PB13 // SPI MISO

#define BEEPER_PIN PC6 // LCD_ENC

#elif ENABLED(VET6_12864_LCD)

/* VET6 12864 LCD */
#define LCD_PINS_RS PA4
#define LCD_PINS_EN PA7
#define LCD_PINS_D4 PA5

#define BTN_ENC PC5
#define BTN_EN1 PB10
#define BTN_EN2 PA6

#elif ENABLED(DWIN_CREALITY_LCD)

// RET6 DWIN ENCODER LCD
#define BTN_ENC PB14
#define BTN_EN1 PB15
#define BTN_EN2 PB12

// #define LCD_LED_PIN                     PB2
#ifndef BEEPER_PIN
#define BEEPER_PIN PB13
#undef SPEAKER
#endif

#elif ENABLED(DWIN_VET6_CREALITY_LCD)

// VET6 DWIN ENCODER LCD
#define BTN_ENC PA6
#define BTN_EN1 PA7
#define BTN_EN2 PA4

#define BEEPER_PIN PA5

#endif
`

I currently don't have time to find out what was wrong to modify the pins file in the repository. Maybe I will find time at some point, or you have time to do it?

jam-berg, 2024-05-09:
Fixed and created a pull request on the main repo. LCD_PINS_RS was not defined

closed 2024-05-09 by jam-berg

shadow578, 2024-05-10:
Hi there,

seems like i'm a bit late to this issue :).

as you've already found out the CR10 display (or whatever it's actually called) is indeed supported on HC32, so the most likely cause of the issue is (as you've also already found) that the pins are defined incorrectly.